devanlai/dap42 #3: Implement USB CDC control lines (RTS, DTR)

kaofishy, 2017-04-24:
Are there any plans to implement hardware control lines for the USB to serial port on dap42? I looked over the source and it seems to only require adding a callback to cdc_setup(), or am I missing something? Having controllable RTS and DTR is very useful when programming an MCU over serial bootloader as they can be used to reset and/or put the MCU into programming mode.

(Great project BTW, I've found it very useful when I want a quick and dirty ARM debugger and USB serial port)

devanlai, 2017-04-24:
I'm not too likely to add support for hardware control lines since I don't normally use them, though I can see the appeal of having GPIO that can be controlled from a serial API.

If you want to try doing it yourself, it should be a simple as you say - registering a callback in `cdc_setup()` to allow the host to set the DTR/RTS state and de-asserting DTR/RTS in `cdc_uart_app_reset()` when the host gracefully closes the connection.

Thanks for the feedback - I use my boards for exactly the same thing - it's great to hear that someone else is able to make use of it.

kaofishy, 2017-04-25:
Thanks, I'll play around with the code and if I don't butcher it too much I may submit a pull request.