devanlai/dap42 #4: Can not support M7 kernel KV56

neomissing, 2017-09-23:
I tested DAP42 debug KV56, found unable to recognize the M7 kernel chip, is not the firmware does not support？

devanlai, 2017-09-23:
I suspect that the issue is with the host software, since the KV56 looks like a relatively new part, but I need more information to rule out a hardware/firmware issue.

Could you provide a little more detail about what software you're using (ie, Keil, openocd, IAR) and what the exact error is, word for word? Also, if you're using a publicly available development board, could you provide a link to the documentation?

neomissing, 2017-09-24:
Thank you, I have resolved, is the MDK reset configuration option problem, you can identify the download

closed 2017-09-24 by devanlai