devanlai/dap42 #7: Adding full JTAG: what does this mean?

xcvista, 2018-11-21:
I was attempting to add full JTAG into this,however I got this from `openocd -d`:

```
Debug: 432 3354 adi_v5_jtag.c:572 jtagdp_transaction_endcheck(): jtag-dp: CTRL/STAT 0xf0000021
Error: 433 3354 adi_v5_jtag.c:580 jtagdp_transaction_endcheck(): JTAG-DP STICKY ERROR
Debug: 434 3356 target.c:1561 target_call_event_callbacks(): target event 0 (gdb-halt)
User : 435 3356 target.c:2740 handle_target(): Polling target stm32f103cb.cpu failed, trying to reexamine
Debug: 436 3356 target.c:1561 target_call_event_callbacks(): target event 21 (examine-start)
Debug: 437 3356 arm_adi_v5.c:603 dap_dp_init():
Debug: 438 3360 arm_adi_v5.c:637 dap_dp_init(): DAP: wait CDBGPWRUPACK
Debug: 439 3360 arm_adi_v5.h:428 dap_dp_poll_register(): DAP: poll 4, mask 0x20000000, value 0x20000000
Debug: 440 3366 arm_adi_v5.c:644 dap_dp_init(): DAP: wait CSYSPWRUPACK
Debug: 441 3366 arm_adi_v5.h:428 dap_dp_poll_register(): DAP: poll 4, mask 0x80000000, value 0x80000000
Debug: 442 3382 arm_adi_v5.c:785 dap_find_ap(): Found AHB-AP at AP index: 0 (IDR=0x14770011)
Debug: 443 3393 arm_adi_v5.c:712 mem_ap_init(): MEM_AP Packed Transfers: enabled
Debug: 444 3393 arm_adi_v5.c:723 mem_ap_init(): MEM_AP CFG: large data 0, long address 0, big-endian 0
Debug: 445 3401 adi_v5_jtag.c:572 jtagdp_transaction_endcheck(): jtag-dp: CTRL/STAT 0xf0000021
Error: 446 3401 adi_v5_jtag.c:580 jtagdp_transaction_endcheck(): JTAG-DP STICKY ERROR
Error: 447 3409 arm_adi_v5.c:489 mem_ap_read(): Failed to read memory at 0xe000ed04
Debug: 448 3409 target.c:2310 target_read_u32(): address: 0xe000ed00 failed
User : 449 3409 target.c:2748 handle_target(): Examination failed, GDB will be halted. Polling again in 3100ms
```

What is going on?

devanlai, 2018-11-21:
I have not tested JTAG support recently, so I can't offer much advice right now. If I were testing this again, I would try the following next steps:

1. Increase the openocd debug levels to verbose (`-d4`). It's probably a torrent of overly specific log info, but it might show something interesting.
2. Compare the logs using openocd with a known good JTAG adapter (preferably one that gives openocd full control, like an FT232H as opposed to a high-level adapter like the ST-Link) at the same nominal clock rate.
3. As above, but also take a logic analyzer trace and decode it. I believe Sigrok/Pulseview have JTAG/SWD decoders.

xcvista, 2018-11-21:
I don't have that type of logic analyzer... Will ULink 2 make a good reference adapter, since it with the latest firmware also uses CMSIS-DAP?

devanlai, 2018-11-21:
Sigrok supports a [fairly broad range](https://sigrok.org/wiki/Supported_hardware) of USB-based logic analyzers, but I understand if you don't want to add a software configuration problem on top of your hardware debugging problem.

The ULink2 seems like it would be a good reference from what you've described.