devanlai/dap42 #8: [FR] add JTAG support for STM32F103

ildar, 2019-09-12:
I successfully use dap42 on a general STM32F103 board (BluePill exactly).
I guess nothing prevents JTAG be on for it. Thanks.

devanlai, 2019-09-12:
Hi, it is technically possible to add JTAG support, but I haven't spent much time integrating it because I usually don't need full JTAG and from the last time I looked into it, the performance was not great. This is in part because the USB handling is not particularly optimized, and in part because CMSIS-DAP 1.0 is not very fast as far as JTAG transports go.

ildar, 2019-11-18:
probably related: https://github.com/devanlai/dap42-hardware/issues/2

xobs, 2025-02-26:
I have initial JTAG support in a branch: https://github.com/xobs/dap42/tree/add-jtag-support

This has been tested on STM32F042 hardware, and I have a new PCB coming that fixes the hardware issues. I'll make a PR once that's actually tested and working.