devanlai/dap42 #11: can't work on bluepill

kabirz, 2020-05-18:
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 1.0
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
in procedure 'program' 
in procedure 'init' called at file "embedded:startup.tcl", line 506
in procedure 'ocd_bouncer'
** OpenOCD init failed **
shutdown command invoked

devanlai, 2020-05-18:
Hi @kabirz. Debugging debug hardware issues can be a tricky, but I'll do my best to help you out. If you could answer the following questions, it would help us both figure out what's going on:

- What hardware are you running the DAP42 firmware on (native STM32F042 hardware, STLink/v2 clone hardware, a development board like the bluepill)
- What target hardware are you connecting to?
- What configuration files / commands did you pass to openocd?

Additionally, the default openocd logging doesn't give enough of a hint as to what's going wrong in this case. Please pass in the `-d3` option to openocd to increase the debug log level and attach the updated log output.

kabirz, 2020-05-18:
> Hi @kabirz. Debugging debug hardware issues can be a tricky, but I'll do my best to help you out. If you could answer the following questions, it would help us both figure out what's going on: [...]

1. hardware: BluePill (mcu: stm32f103 c8t6)
2. I have two BluePill, one for dap board ,one for target board.
3. cmd: openocd -f interface/cmsis-dap.cfg -f target/stm32f1x.cfg -c "program build/led.bin 0x08000000 verify reset exit"

devanlai, 2020-05-18:
Hi @kabirz, I dug up a couple of my bluepill boards and flashed the latest 1.31 firmware image (using [DAP103-dapboot-combined-stlink.bin](https://github.com/devanlai/dap42/releases/download/v1.31/DAP103-dapboot-combined-stlink.bin).

I ran your command with a test firmware bin I had lying around and it was able to flash the target bluepill just fine. Could you confirm that your wiring is correct? There should be three to four connections, depending on whether you're powering the target independently or from the debug bluepill:

- PB14 on the debugger to SWDIO on the target
- PB13 on the debugger to SWCLK on the target
- GND should be connected on both bluepills
- 3.3V should be connected to 3.3V if powering from the debugger, otherwise the target should be powered when flashing it.

Can you confirm that your wiring checks out?

kabirz, 2020-05-18:
@devanlai, it seams my wiring issue, I tried several times, finally it's worked.
And can you add a macro to select remap SWDIO/SWCLK to PA13/PA14? it seams better for connecting debugger and target.

devanlai, 2020-05-19:
Hi @kabirz ,

I'm glad you were able to resolve the issue.
I'll look into adding another target that uses the bluepill's own SWDIO/SWCLK pins as the debugger SWDIO/SWCLK pins. It isn't technically difficult to do - it's mostly a matter of deciding what to name the configuration.

closed 2020-05-19 by devanlai