devanlai/dap42 #12: Using 4 pin front header on Blue Pill

verjus, 2021-01-31:
Hi, 

This is not really an issue but I am trying to use dap42 on an Blue Pill board with no side headers. Therefore, I would like to use the 4 pin front header (labeled swdio and swclk). I tried to redefine the GPIO PINs to use them instead of PB13/14 by modifying the file src/stm32f103/bluepill/DAP/CMSIS_DAP_config.h as follows:

From:

```
#define SWCLK_GPIO_PORT         GPIOB
#define SWCLK_GPIO_PIN          GPIO13
#define SWDIO_GPIO_PORT         GPIOB
#define SWDIO_GPIO_PIN          GPIO14
#define nRESET_GPIO_PORT        GPIOB
#define nRESET_GPIO_PIN         GPIO0

#define LED_CON_GPIO_PORT       GPIOC
#define LED_CON_GPIO_PIN        GPIO13
#define LED_RUN_GPIO_PORT       GPIOC
#define LED_RUN_GPIO_PIN        GPIO13
#define LED_ACT_GPIO_PORT       GPIOC
#define LED_ACT_GPIO_PIN        GPIO13

#define SWDIO_GPIO_PIN_NUM      14

```
TO:

```
#define SWCLK_GPIO_PORT         GPIOA
#define SWCLK_GPIO_PIN          GPIO14
#define SWDIO_GPIO_PORT         GPIOA
#define SWDIO_GPIO_PIN          GPIO13
#define nRESET_GPIO_PORT        GPIOB
#define nRESET_GPIO_PIN         GPIO0

#define LED_CON_GPIO_PORT       GPIOC
#define LED_CON_GPIO_PIN        GPIO13
#define LED_RUN_GPIO_PORT       GPIOC
#define LED_RUN_GPIO_PIN        GPIO13
#define LED_ACT_GPIO_PORT       GPIOC
#define LED_ACT_GPIO_PIN        GPIO13

#define SWDIO_GPIO_PIN_NUM      13

```

But it's not working. When I try, I get:

```
openocd -f /data2/dap42/openocd/interface-dap42.cfg -c 'transport select swd' -f target/nrf52.cfg
Open On-Chip Debugger 0.11.0-rc2+dev-00002-g427552c-dirty (2021-01-31-10:09)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
swd
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 1.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Error: Error connecting DP: cannot read IDR

```

What am I missing?

devanlai, 2021-02-02:
Hi @verjus, the SWD and JTAG pins require special remapping on STM32F1xx parts before they can be reused.

You can see the documentation in the libopencm3 docs for this on the [`gpio_primary_remap()`](http://libopencm3.org/docs/latest/stm32f1/html/group__gpio__defines.html#ga9fa7b8ed1d38d8b3d0e22bdfbabb9003) function.

In `gpio_setup()` in [`stm32f103/bluepill/target.c`](https://github.com/devanlai/dap42/blob/master/src/stm32f103/bluepill/target.c), you'll want to enable the AFIO clock and then remap the SWD pins, e.g:
```c
    /* Remap SWD pins as GPIO (so we can drive them) */
    rcc_periph_clock_enable(RCC_AFIO);
    gpio_primary_remap(AFIO_MAPR_SWJ_CFG_JTAG_OFF_SW_OFF, 0);
```

From my testing, that seems to be sufficient to use the 4-pin SWD header.

verjus, 2021-02-02:
Thank you, @devanlai. I followed your instructions and modified the function as follows:

```
void gpio_setup(void) {
    /*
      LED0, 1, 2 on PC13
      TX, RX (MCU-side) on PA2, PA3
      TGT_RST on PB0
      TGT_SWDIO, TGT_SWCLK on PB14, PB13
      TGT_SWO on PB11
    */

    /* Enable GPIOA, GPIOB and GPIOC clocks. */
    rcc_periph_clock_enable(RCC_GPIOA);
    rcc_periph_clock_enable(RCC_GPIOB);
    rcc_periph_clock_enable(RCC_GPIOC);

#ifndef USE_SWD
    /* Enable alternate function io clock */
    rcc_periph_clock_enable(RCC_AFIO);
    /* Disable SWD and JTAG to allow full use of the ports PA13, PA14, PA15 */
    gpio_primary_remap(AFIO_MAPR_SWJ_CFG_JTAG_OFF_SW_OFF,0);
#endif
    
    /* Setup LEDs as open-drain outputs */
    const uint8_t mode = GPIO_MODE_OUTPUT_10_MHZ;
    const uint8_t conf = (LED_OPEN_DRAIN ? GPIO_CNF_OUTPUT_OPENDRAIN
                                         : GPIO_CNF_OUTPUT_PUSHPULL);
    gpio_set_mode(GPIOC, mode, conf, GPIO13);
}

```
and it now works perfectly. Thanks again for your help! Let me know if you would like me to create a pull request with a  #ifndef USE_SWD macro, as above. It seems like I will be able to disable APPROTECT on nrf52 with nrf52.dap apreg 1 0x04 0x01.