devanlai/dap42 #21: Unable to flash stlink v2 clone

atkinson, 2024-08-16:
Hello,

Following the instructions here: https://github.com/devanlai/dap42/blob/master/FLASHING.md#dap103-stlinkv2-clone-dongle

I'm not able to flash my stlink v2 clone. I've tried both mac os and linux with the same result.

```
> openocd -f interface/cmsis-dap.cfg -f target/stm32f1x.cfg \
    -c "init; reset halt" \
    -c "stm32f1x unlock 0; reset halt" \
    -c "flash erase_sector 0 0 last" \
    -c "program DAP103-dapboot-combined-stlink.bin verify reset exit 0x08000000"

Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Error: unable to find a matching CMSIS-DAP device

> lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 8087:0025 Intel Corp. Wireless-AC 9260 Bluetooth Adapter
Bus 002 Device 004: ID 046d:c542 Logitech, Inc. M185 compact wireless mouse
Bus 002 Device 005: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub             
Bus 002 Device 009: ID 0483:3748 STMicroelectronics ST-LINK/V2
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub
```

any help very much appreciated.

devanlai, 2024-08-16:
Hi @atkinson, do you have a second debugger hooked up to flash your clone? From your lsusb, it looks like you only have the target clone connected.

It would be simplest if you have a second debugger wired up to the clone's internal SWD pads (it doesn't have to use CMSIS-DAP; it can be an unmodified STLink clone or any SWD debugger supported by openocd).

Otherwise it's theoretically possible to get a custom firmware payload uploaded through the bootloader that is normally used to upgrade STLink firmware, but it's kind of a pain and I don't have the steps in front of me.

atkinson, 2024-08-19:
Thanks for your advice. I have now setup a pico pi as a cmsis-dap probe and validated that it works using a second pico pi.

I'm now using my pico pi debugger to try and flash the st link v2 clone.

I've removed the case and connected the internal pads to my pico pi debugger, and it all seems good - until I try and talk to it.

```
> openocd -f interface/cmsis-dap.cfg -f target/stm32f1x.cfg -c "init"

Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E66178C1274F4F2C
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x8fe0002f
```

... all looks OK, until I turn on debugging (-d3), and I see that the init command failed.

```
...
Info : 359 376 adi_v5_swd.c:370 swd_connect_single(): SWD DPIDR 0x57020bff
Debug: 360 376 cmsis_dap.c:873 cmsis_dap_swd_read_process(): SWD ack not OK @ 0 JUNK
Debug: 361 376 command.c:545 run_command(): Command 'dap init' failed with error code -4
User : 362 376 command.c:608 command_run_line(): 
Debug: 363 376 command.c:545 run_command(): Command 'init' failed with error code -4
User : 364 376 command.c:608 command_run_line(): 
Debug: 365 376 target.c:2199 target_free_all_working_areas_restore(): freeing all working areas
Debug: 366 377 cmsis_dap.c:1110 cmsis_dap_swd_switch_seq(): SWD-to-JTAG
```

[Full debug output here](https://gist.github.com/atkinson/1c25be0b9675f6bf28f258eff757fed8)

If I had hair I would have been pulling it out. I realise this isn't a dap42 issue, but do you have any suggestions?

gratefully
Rich

devanlai, 2024-08-19:
Hmm, I seem to have misplaced my pico board, so I can't recreate your setup exactly. I would generally suggest making sure that you're connected to the correct pads (the internal test point positioning on the clones vary, so you may have to follow the traces to confirm you've got SWDIO and SWCLK wired correctly).

Additionally just as a sanity check, make sure the STLink is powered; I normally plug mine into the same USB hub I'm powering the debugger from to reduce the number of test points I have to trace.

atkinson, 2024-08-22:
Thanks for your help, really appreciated. I couldn't get to the bottom of it and eventually bought a proper pico probe.

thanks again!

closed 2024-08-22 by atkinson