devanlai/dap42 #24: CDC uart not working, when flashed on nucleo

omicronns, 2025-02-28:
I flashed this firmware (DAP103.bin) onto nucleo debugger, and I'm not getting uart output via cdc interface on my PC. I can confirm, that uart signal is reaching MCU pin, and pin configuration seem to match actual connections on the board. I think part of the problem is https://github.com/devanlai/dap42/blob/e9fdd1180806f8591a23013ec0ecedb509517a1d/src/stm32f103/stlinkv2-1/config.h#L36 where other console related parameters are USART2, but here it is set to USART1. Unfortunately changing this parameter alone doesn't fix the problem.

devanlai, 2025-03-01:
You're right that it shouldn't be using `CONSOLE_USART1`, but you may also be using the wrong target for your hardware. The `DAP103.bin` is built using the `STLINKV2-DONGLE` target and targets STLink/v2 "mini" clone hardware, which only breaks out a subset of the full STLink/v2 pins. Because of that, I decided to repurpose the SWIM pin on USART3 as an RX pin, rather than using USART2 which isn't routed to anything accessible.

The `STLINKV2-1-STBOOT` target, on the other hand, targets real embedded STLink/v2-1s and probably would work if fixed to use correctly select USART2, but I don't think I made any official releases for that target because it was designed to be built into an STLink firmware update package that could be installed using ST's own updater tool. It also requires some tools and files I'd have to pull off an old laptop.

So if you've flashed DAP103 firmware, it would be trying to receive on USART3, rather than using USART2.

If you want to continue down this path, it would probably be easiest to modify the [config.h](../blob/master/src/stm32f103/stlinkv2-dongle/config.h) for the `STLINKV2-DONGLE` target to use the UART settings from the `STLINKV2-1-STBOOT` config that you're looking at and fix `CONSOLE_USART`.

omicronns, 2025-03-01:
Right, I mixed the targets, I tried NUCLEO-STBOOT target later, but it did not start correctly, I assume because I don't have bootloader on the MCU. I managed to make NUCLEO target with no bootloader support, but there was another piece of code that was obstructing correct boot. I will create a pull request.

omicronns, 2025-03-01:
https://github.com/devanlai/dap42/pull/25

devanlai, 2025-03-01:
Thanks, I've merged your PR with the USART fix and the new non-bootloader target.

closed 2025-03-01 by devanlai